PX4/px4_ros_com #107: Offboard control documentation not up to date

vranki, 2021-08-20:
I'm still struggling with this week's changes and my drone project (using c++ offboard control) is frozen.

I'm trying to check my setup with the PX4 documentation and examples.

https://docs.px4.io/master/en/ros/ros2_offboard_control.html

This page mentions modifications to PX4-Autopilot/msg/tools/uorb_rtps_message_ids.yaml - but this file doesn't exist.
Same with px4_ros_com/templates/uorb_rtps_message_ids.yaml
 
Probably these files have changed in some update. It would be good if someone could go through the documentation example and check it's up to date. As a end-user it is very difficult to fix the environment currently when the documentation doesn't match the implementation.

TSC21, 2021-08-20:
Sorry for thr inconvinience. I will update the docs in the following weeks. No prediction though.

Meanwhile can you please explain what's yout problem atm? The docs should not matter in this case to make the offboard example to work. I will keep in mind this issue though: https://github.com/PX4/px4_ros_com/issues/106#issue-974183548. This is the blocker currently to make it work.

vranki, 2021-08-23:
I ran the documented setup with up to date repos.

Some notes:
```
pxh> micrortps_client start -t UDP 
```
This step is not needed, at least in my setup the micrortps_client is already running.

```
ros2 topic hz /SensorCombined_PubSubTopic
```
Fails. This should be ```ros2 topic hz /fmu/sensor_combined/out```

My next sanity test was to build the example here: https://docs.px4.io/master/en/middleware/micrortps.html#creating-a-fast-dds-listener-application

After copypasting the function content, the build fails:

sensor_combinedSubscriber.cxx:87:5: error: ‘sensor_combined_’ was not declared in this scope; did you mean ‘sensor_combined’?

sensor_combined_Subscriber.cxx should be sensor_combinedSubscriber.cxx (note underscore difference)

There are lot of changed names that need to be corrected. Also magnetometer/baro values do not exist so I commented them out.

```
sensor_combinedSubscriber.cxx: In member function ‘virtual void sensor_combinedSubscriber::SubListener::onNewDataMessage(eprosima::fastrtps::Subscriber*)’:
sensor_combinedSubscriber.cxx:101:53: error: ‘class sensor_combined’ has no member named ‘gyro_integral_dt’; did you mean ‘gyro_integral_dt_’?
  101 |             std::cout << "gyro_integral_dt: " << st.gyro_integral_dt() << std::endl;
      |                                                     ^~~~~~~~~~~~~~~~
      |                                                     gyro_integral_dt_
sensor_combinedSubscriber.cxx:102:69: error: ‘class sensor_combined’ has no member named ‘accelerometer_timestamp_relative’; did you mean ‘accelerometer_timestamp_relative_’?
  102 |             std::cout << "accelerometer_timestamp_relative: " << st.accelerometer_timestamp_relative() << std::endl;
      |                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                     accelerometer_timestamp_relative_
sensor_combinedSubscriber.cxx:106:62: error: ‘class sensor_combined’ has no member named ‘accelerometer_integral_dt’; did you mean ‘accelerometer_integral_dt_’?
  106 |             std::cout << "accelerometer_integral_dt: " << st.accelerometer_integral_dt() << std::endl;
      |                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                              accelerometer_integral_dt_
sensor_combinedSubscriber.cxx:107:68: error: ‘class sensor_combined’ has no member named ‘magnetometer_timestamp_relative’; did you mean ‘accelerometer_timestamp_relative_’?
  107 |             std::cout << "magnetometer_timestamp_relative: " << st.magnetometer_timestamp_relative() << std::endl;
      |                                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                    accelerometer_timestamp_relative_
sensor_combinedSubscriber.cxx:108:52: error: ‘class sensor_combined’ has no member named ‘magnetometer_ga’
  108 |             std::cout << "magnetometer_ga: " << st.magnetometer_ga().at(0);
      |                                                    ^~~~~~~~~~~~~~~
sensor_combinedSubscriber.cxx:109:37: error: ‘class sensor_combined’ has no member named ‘magnetometer_ga’
  109 |             std::cout << ", " << st.magnetometer_ga().at(1);
      |                                     ^~~~~~~~~~~~~~~
sensor_combinedSubscriber.cxx:110:37: error: ‘class sensor_combined’ has no member named ‘magnetometer_ga’
  110 |             std::cout << ", " << st.magnetometer_ga().at(2) << std::endl;
      |                                     ^~~~~~~~~~~~~~~
sensor_combinedSubscriber.cxx:111:60: error: ‘class sensor_combined’ has no member named ‘baro_timestamp_relative’
  111 |             std::cout << "baro_timestamp_relative: " << st.baro_timestamp_relative() << std::endl;
      |                                                            ^~~~~~~~~~~~~~~~~~~~~~~
sensor_combinedSubscriber.cxx:112:51: error: ‘class sensor_combined’ has no member named ‘baro_alt_meter’
  112 |             std::cout << "baro_alt_meter: " << st.baro_alt_meter() << std::endl;
      |                                                   ^~~~~~~~~~~~~~
sensor_combinedSubscriber.cxx:113:54: error: ‘class sensor_combined’ has no member named ‘baro_temp_celcius’
  113 |             std::cout << "baro_temp_celcius: " << st.baro_temp_celcius() << std::endl;
      |                                                      ^~~~~~~~~~~~~~~~~
make: *** [makefile_x64Linux2.6gcc:51: output/x64Linux2.6gcc/sensor_combinedSubscriber.o] Error 1
```

The binary file to run has changed: bin/*/sensor_combined_PublisherSubscriber subscriber -> bin/*/sensor_combinedPublisherSubscriber subscriber

The example seems to have old topic name, I changed Rparam.topic.topicName = "/fmu/sensor_combined/out"; but even after this change the example doesn't print any data. Client is running and the python sensor_combined_listener.launch.py still works. 

I'll try the c++ offboard control example next and see if it work. I'll write another comment about it.

TSC21, 2021-08-23:
> I ran the documented setup with up to date repos.
>
> Some notes:
>
> ```
> pxh> micrortps_client start -t UDP
> ```
>
> This step is not needed, at least in my setup the micrortps_client is already running. [...]

@vranki the documentation is being updated, but what you are describing above is a complete different problem, probably because you have the wrong setup on your system. The topic names and type naming that get generated from the templates depend on a specific FastRTPS version being used or if you are using the ROS2 or non-ROS2 version. You are using the non-ROS2 version, and we clearly state on the docs that there's a difference between them (and what you are testing is the agent generated from the PX4-Autopilot build + a sourced ROS2 environment, which in the end results on the build errors you are encountering).

So please I ask you to get back to the ROS2 docs as documented in https://docs.px4.io/master/en/ros/ros2_comm.html and use the workflow documented there. The only difference there that should be accounted is the topic naming. The rest is the same.
I am currently on holiday so my availability to go over this is short. Sorry for any inconvenience.

vranki, 2021-08-23:
Thank you for the answer - I'm still quite beginner with ROS2 and probably followed wrong leads. I hope the documentation fixes are useful.

I managed to add the topics I need and got my c++ offboard project working again, so the issues I've experienced last week are now fixed. I'll leave the issue open until the remaining documentation issues are fixed.

TSC21, 2021-09-02:
Documentation got updated in https://github.com/PX4/PX4-user_guide/pull/1522. Closign.

closed 2021-09-02 by TSC21