PX4/px4_ros_com #108: Cannot turn on additional ros topics

nfry321, 2021-08-20:
Related to #107 
I tried to follow the procedures mentioned in #99 but modified for the changes that have happened recently but PX4 firmware doesn't build and gives this error when running `make px4_sitl_rtps gazebo`:

```
src/modules/micrortps_bridge/micrortps_client/microRTPS_client.cpp:319: error: undefined reference to 'serialize_vehicle_local_position(ucdrBuffer*, vehicle_local_position_s const*, char*, unsigned int*)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make: *** [Makefile:225: px4_sitl_rtps] Error 1
```

## Steps to reproduce
Pulled latest versions of `px4_msgs`, `px4_ros_com` and `PX4_Autopilot`.

In `/PX4-Autopilot/msg/tools/urtps_bridge_topics.yaml`

Added 

```bash
- msg:     vehicle_local_position
  send:    true
```

Then use `msg/tools/uorb_to_ros_urtps_topics.py` script to regenerate the file under `<ros2_workspace>/src/px4_ros_com/templates/` specifying the input [-i] and output [-o] files:

```bash
./uorb_to_ros_urtps_topics.py -i urtps_bridge_topics.yaml -o ~/ros2_ws/src/px4_ros_com/templates/urtps_bridge_topics.yaml
```

I checked the output and it is as expected 
```
  - msg: VehicleLocalPosition
    send: true
```

Use the scripts `px4_ros_com/scripts/clean_all.bash` and `px4_ros_com/scripts/build_ros2_workspace.bash` to rebuild my ros workspace. Sourced this and tried `make px4_sitl_rtps gazebo`.

Any help is appreciated. The local position was previously published by default and I have used it in a script, which I am now converting to use the changed topic names.

TSC21, 2021-08-20:
Did you make clean and built the PX4-Autopilot side?

nfry321, 2021-08-20:
I had done `make clean` and got the same error. To be thorough, just now I manually removed the `PX4_Autopilot/build/px4_sitl_rtps` directory and rebuilt, and it has worked, so it seems like clean wasn't fully removing something. Thanks for the help.

LorenzMeier, 2021-08-21:
@TSC21 Could you add a commit to make sure that directory is correctly built iteratively and also cleaned correctly?

@dagar FYI

dagar, 2021-08-21:
@nfry321 the PX4-Autopilot dependency should be fixed in https://github.com/PX4/PX4-Autopilot/pull/18120.

TSC21, 2021-08-21:
> @nfry321 the PX4-Autopilot dependency should be fixed in [PX4/PX4-Autopilot#18120](https://github.com/PX4/PX4-Autopilot/pull/18120).

The same dependency needs to be added on the CMakeLists.txt in this repo.