PX4/px4_ros_com #114: Possible issue with 'generate_microRTPS_bridge.py' when ROS2 is built from source.

rblilja, 2021-10-02:
If ROS2 has been built from source, I suspect line 210 and 211 in 'generate_microRTPS_bridge.py' is bound to fail: 

`fastrtps_version = re.search(r'Version:\s*([\dd.]+)', subprocess.check_output(
        "dpkg -s ros-" + ros2_distro + "-fastrtps 2>/dev/null | grep -i version", shell=True).decode("utf-8").strip()).group(1)`

Works on a machine with Foxy installed from Debian packages. But not on another machine with Foxy built from source (above lines fail). Perhaps I'm missing something? Also tried building and installing FastDDS 2.0.1 (as well as 1.8.2) from source separately (shall already be included with Foxy).

MJ-Graf, 2021-11-20:
Same here, I built ros2-foxy from source on ubuntu. Thereby, fastrtps is already part of the installation but cannot be detected using dpkg since it wasn't obtained from any ppa.

tank104, 2022-04-05:
How did you fix this issue?

Is this the error that you get:
andrew@ANDREW-LAPTOP:~/PX4-Autopilot$ make px4_sitl_rtps
[27/312] Generating RTPS topic bridge
openjdk version "13.0.7" 2021-04-20
OpenJDK Runtime Environment (build 13.0.7+5-Ubuntu-0ubuntu120.04)
OpenJDK 64-Bit Server VM (build 13.0.7+5-Ubuntu-0ubuntu120.04, mixed mode, sharing)
Traceback (most recent call last):
  **File "/home/andrew/PX4-Autopilot/msg/tools/generate_microRTPS_bridge.py", line 206, in <module>
    fastrtps_version = subprocess.check_output(**
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'ldconfig -v 2>/dev/null | grep libfastrtps' returned non-zero exit status 1.

[28/307] Building CXX object src/modules/microdds_cl...s/modules__microdds_client.dir/microdds_client.cpp.o
FAILED: src/modules/microdds_client/CMakeFiles/modules__microdds_client.dir/microdds_client.cpp.o
/usr/bin/c++  -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"microdds_client\" -DPX4_MAIN=microdds_client_app_main -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/px4/sitl/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../platforms/posix/include -Iexternal/Install/include -Isrc/modules/microdds_client/include -O2 -g -DNDEBUG -fPIC   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -fno-rtti -D__KERNEL__ -std=gnu++14 -MD -MT src/modules/microdds_client/CMakeFiles/modules__microdds_client.dir/microdds_client.cpp.o -MF src/modules/microdds_client/CMakeFiles/modules__microdds_client.dir/microdds_client.cpp.o.d -o src/modules/microdds_client/CMakeFiles/modules__microdds_client.dir/microdds_client.cpp.o -c ../../src/modules/microdds_client/microdds_client.cpp
In file included from ../../src/modules/microdds_client/microdds_client.cpp:38:
../../src/modules/microdds_client/microdds_client.h:38:10: fatal error: src/modules/micrortps_bridge/micrortps_client/dds_topics.h: No such file or directory
   38 | #include <src/modules/micrortps_bridge/micrortps_client/dds_topics.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

tank104, 2022-04-05:
I got it sorted by running this:
echo "source /opt/ros/foxy/setup.bash" >> ~/.bashrc
source /opt/ros/foxy/setup.bash

gorghino, 2022-12-27:
I got the same error with ROS2 Humble built with sources.

I'm already sourcing humble with
```. ~/ros2_humble/install/local_setup.bash```

but then I get

```gorgo@razer:~/PX4-Autopilot$ make mro_ctrl-zero-h7-oem_rtps
-- PX4 version: v1.13.2-3-g985482badb
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- PX4 config file: /home/gorgo/PX4-Autopilot/boards/mro/ctrl-zero-h7-oem/rtps.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m7
-- ROMFSROOT px4fmu_common
-- SERIAL_GPS1 /dev/ttyS2
-- SERIAL_TEL1 /dev/ttyS0
-- SERIAL_TEL2 /dev/ttyS1
-- SERIAL_TEL3 /dev/ttyS4
-- UAVCAN_INTERFACES 2
-- PX4 config: mro_ctrl-zero-h7-oem_rtps
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 9.3.1
-- The C compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Enabling double FP precision hardware instructions
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Using C++11
-- Release build type: MinSizeRel
-- fastrtpsgen version 1.0.4
-- microRTPS bridge:
--   Publish to the bridge from: collision_constraints, sensor_combined, timesync, timesync_status, trajectory_waypoint, vehicle_control_mode, vehicle_odometry, vehicle_status, vehicle_trajectory_waypoint_desired
--   Subscribe from the bridge to: debug_array, debug_key_value, debug_value, debug_vect, offboard_control_mode, onboard_computer_status, optical_flow, position_setpoint, position_setpoint_triplet, telemetry_status, timesync, trajectory_bezier, vehicle_command, vehicle_local_position_setpoint, vehicle_trajectory_bezier, vehicle_trajectory_waypoint, trajectory_setpoint, vehicle_mocap_odometry, vehicle_visual_odometry
-- ROMFS: ROMFS/px4fmu_common
-- ROMFS:  Adding platforms/nuttx/init/stm32h7/rc.board_arch_defaults -> /etc/init.d/rc.board_arch_defaults
-- ROMFS:  Adding boards/mro/ctrl-zero-h7-oem/init/rc.board_defaults -> /etc/init.d/rc.board_defaults
-- ROMFS:  Adding boards/mro/ctrl-zero-h7-oem/init/rc.board_sensors -> /etc/init.d/rc.board_sensors
-- ROMFS:  Adding boards/mro/ctrl-zero-h7-oem/extras/mro_ctrl-zero-h7-oem_bootloader.bin -> /etc/extras/mro_ctrl-zero-h7-oem_bootloader.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gorgo/PX4-Autopilot/build/mro_ctrl-zero-h7-oem_rtps
[0/1123] git submodule src/modules/microdds_client/Micro-XRCE-DDS-Client
[1/1123] git submodule platforms/nuttx/NuttX/nuttx
[5/1123] git submodule src/drivers/gps/devices
[12/1123] git submodule src/drivers/uavcan/libuavcan
[15/1123] git submodule src/modules/mavlink/mavlink
[16/1123] git submodule src/modules/micrortps_bridge/micro-CDR
[17/1123] Generating Mavlink development: src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /home/gorgo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Parsing /home/gorgo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /home/gorgo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Parsing /home/gorgo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Validating /home/gorgo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Parsing /home/gorgo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Validating /home/gorgo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
[... 1 lines cut ...]
Merged enum MAV_CMD
Merged enum MAV_CMD
Found 229 MAVLink message types in 4 XML files
Generating C implementation in directory /home/gorgo/PX4-Autopilot/build/mro_ctrl-zero-h7-oem_rtps/mavlink/development
Generating C implementation in directory /home/gorgo/PX4-Autopilot/build/mro_ctrl-zero-h7-oem_rtps/mavlink/standard
Generating C implementation in directory /home/gorgo/PX4-Autopilot/build/mro_ctrl-zero-h7-oem_rtps/mavlink/common
Generating C implementation in directory /home/gorgo/PX4-Autopilot/build/mro_ctrl-zero-h7-oem_rtps/mavlink/minimal
Copying fixed headers for protocol 2.0 to /home/gorgo/PX4-Autopilot/build/mro_ctrl-zero-h7-oem_rtps/mavlink
[18/1123] Generating RTPS topic bridge
openjdk version "13.0.7" 2021-04-20
OpenJDK Runtime Environment (build 13.0.7+5-Ubuntu-0ubuntu120.04)
OpenJDK 64-Bit Server VM (build 13.0.7+5-Ubuntu-0ubuntu120.04, mixed mode)
Traceback (most recent call last):
  File "/home/gorgo/PX4-Autopilot/msg/tools/generate_microRTPS_bridge.py", line 210, in <module>
    fastrtps_version = re.search(r'Version:\s*([\dd.]+)', subprocess.check_output(
[... cut ...]
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'dpkg -s ros-humble-fastrtps 2>/dev/null | grep -i version' returned non-zero exit status 1.
[105/1123] Performing configure step for 'libmicroxrceddsclient_project'
loading initial cache file /home/gorgo/PX4-Autopilot/build/mro_ctrl-zero-h7-oem_rtps/src/modules/microdds_client/tmp/libmicroxrceddsclient_project-cache-MinSizeRel.cmake
-- The C compiler identification is GNU 9.3.1
....
```

gorghino, 2022-12-28:
I used to source humble before compiling. Without sourcing it seems to compile correctly.