PX4/px4_ros_com #117: offboard_controll working in gazebo, not on pixhawk 4

b0hne, 2021-10-12:
Hi,
I am trying to get the offboard example running on a pixhawk 4 using a nanopi neo 3 as companion computer. The micrortps_bridge is working via the usb port and topics can be echoed.
The whole setup works in gazebo, but not on the actual board.
I noticed that the published timestamps are not in world time, but only count from system start. timesync does not publish anything, so offboard_example publishes timestamp 0. I tried using system time and the timestamp of "fmu/vehicle_status/out". but it dosn't have any effect.
Is there a working drone project i could have a look at to see where i went wrong?

sstroobants, 2021-12-03:
I'd like to add to this that this keeps happening to me. The issue seems to be that for some reason the timesync RTPS topic is not starting with the microRTPS client. (When I do the same with simulation, it does start).
I am trying to find out where and when this topic should be started, but no luck so far. 
I also cannot find anything about this in de documentation sadly.

sstroobants, 2021-12-03:
I see that if I stop all mavlink with `mavlink stop-all` before connecting the rtps bridge, the timesync does get published. 
Why can this be the case? And is there a way to make sure that this is not necessary, because I would prefer to still be using QGroundControl while controlling the drone from an offboard computer.

b0hne, 2021-12-07:
I haven't pursued this further, ill check out your hotfix to see if i can continue my project until this if fixed. Thanks for the update.
What flight controller are you using? I ordered a pixhwk mini to see if it has the same problem.

sstroobants, 2021-12-07:
I have tried it on both a kakute f7 and a pixhawk mini, so I am afraid you will encounter the same thing there...

b0hne, 2021-12-07:
Ok, thanks for the heads up.

sstroobants, 2021-12-08:
No problem! For now I think I will use this workaround and dive deeper in the reason it happens when I have a bit more time. It seems like we have a similar goal in mind with the hardware/software set-up so let me know if we can share some more info to help each other out.

srgbsv, 2022-03-20:
Hello. I have the similar issue. I have connected px4 board to jetson nano via UART port (/dev/ttyTHS1). When I start micrortps_bridge, I receive sensor data from px4 but I can't send message to px4 board. micrortps_agent show, that packet was sended, but client show 0 received packets. And timesync/out topic not working too. Then I disabled mavlink on TELEM1 port (MAV_0_MODE set to 0), and then all worked fine. I connected PX4 board to jetson nano by USB and, use /dev/ttyACM0 for controll by QGroundControll.

P.S. I have two boards: Holybro PX4 and Cube Orange. The same issue on both.

littlefred117, 2023-10-30:
Hi, I am trying to do off board control on a Px4 using a Jetson Nano. I want to connect it using QGroundControl and Pymavlink, however I am unsure how to do this (not using ROS). Does someone have experience doing this?