PX4/px4_ros_com #120: No ROS2 topics via micrortps-bridge published

hsu-ret, 2021-11-15:
Hey everyone,
I'm currently trying to set up a micrortps-connection between a pixhawk 2 cube (black) flight controller (https://docs.px4.io/master/en/flight_controller/pixhawk-2.html) and a Raspberry Pi. The flight controller is running the default Firmware version of fmu_v3 where "micrortpsbridge"-module has been added to the cmake file. The Raspberry Pi is using a 20.04 Ubuntu LTS Server-Edition with ROS2-Foxy. As a connection I use the Tele2-Port of the pixhawk and a usb2uart ftdi-adapter (CP2102 USB UART Board(type A)). 

### Issue
To set up a connection between the Raspberry Pi and the flight controller I start the micrortps_client via Mavlink-Console from Q-GroundControll. I can also call a listener on  any available topic there, which produces an output with a timestamp and some data. 
Starting the micrortps_agent on the RPi I get notifications, that topics are published now. 
Entering `ros2 topic list` in a new console on the RPi, none of the topics are listed.
Also starting the sensorCombined listener-node doesn't give me any data, staying at the stage of "Starting listener node...".
After closing the agent and the client, the notifications on both consoles show me an amount of messages "sent" and "received". The Console-Code can be seen below. 

This leaves me wondering, whether the micrortps-bridge is not working correctly (for this setup?) or messages are correctly exchanged but not published into ros2. Is there anything I'm missing about setting up the connection, that causes this problems or did anyone else already have this problem? Any help or advices are welcome, since I'm pretty new to ROS2.

### Console-Code 

**Output from Pixhawk (Mavlink-Console from QGroundControll):**

NuttShell (NSH) NuttX-10.1.0
nsh> micrortps_client start -d /dev/ttyS2 -b 921600 -t UART
INFO  [micrortps_client] UART transport: device: /dev/ttyS2; baudrate: 921600; poll: 1ms; flow_control: No
nsh> listener sensor_accel 

TOPIC: sensor_accel 3 instances

Instance 0:
 sensor_accel_s
    timestamp: 465679695  (0.007060 seconds ago)
    timestamp_sample: 465679429  (266 us before timestamp)
    device_id: 2359330 (Type: 0x24, SPI:4 (0x00)) 
    x: -0.2087
    y: 0.2006
    z: -9.9163
    temperature: 50.0412
    error_count: 0
    clip_counter: [0, 0, 0]
    samples: 5

Instance 1:
 sensor_accel_s
    timestamp: 465728511  (0.000594 seconds ago)
    timestamp_sample: 465728485  (26 us before timestamp)
    device_id: 1114146 (Type: 0x11, SPI:4 (0x00)) 
    x: 2.4120
    y: -2.8786
    z: -7.9107
    temperature: 45.1250
    error_count: 0
    clip_counter: [0, 0, 0]
    samples: 1

Instance 2:
 sensor_accel_s
    timestamp: 465776512  (0.000456 seconds ago)
    timestamp_sample: 465776239  (273 us before timestamp)
    device_id: 2359306 (Type: 0x24, SPI:1 (0x00)) 
    x: 0.0962
    y: -0.0699
    z: -9.7784
    temperature: 55.3128
    error_count: 0
    clip_counter: [0, 0, 0]
    samples: 5
nsh> 

**Output from Raspberry Pi Agent:**

ubuntu@ubuntu:~$ micrortps_agent -d /dev/ttyUSB0 -b 921600 -t UART
--- MicroRTPS Agent ---
[   micrortps_agent   ] Starting link...
[   micrortps_agent   ] UART transport: device: /dev/ttyUSB0; baudrate: 921600; poll: 1ms; flow_control: No
---   Subscribers   ---
- DebugArray subscriber started
- DebugKeyValue subscriber started
- DebugValue subscriber started
- DebugVect subscriber started
- OffboardControlMode subscriber started
- OpticalFlow subscriber started
- PositionSetpoint subscriber started
- PositionSetpointTriplet subscriber started
- TelemetryStatus subscriber started
- Timesync subscriber started
- VehicleCommand subscriber started
- VehicleLocalPositionSetpoint subscriber started
- VehicleTrajectoryWaypoint subscriber started
- OnboardComputerStatus subscriber started
- TrajectoryBezier subscriber started
- VehicleTrajectoryBezier subscriber started
- TrajectorySetpoint subscriber started
- VehicleMocapOdometry subscriber started
- VehicleVisualOdometry subscriber started
-----------------------

----   Publishers  ----
- Timesync publishers started
- TrajectoryWaypoint publisher started
- VehicleControlMode publisher started
- VehicleOdometry publisher started
- VehicleStatus publisher started
- CollisionConstraints publisher started
- TimesyncStatus publisher started
- SensorCombined publisher started
- VehicleTrajectoryWaypointDesired publisher started
-----------------------
^C
[   micrortps_agent   ] Interrupt signal (2) received.
[ micrortps_transport ] Closed UART.
[   micrortps_agent   ] RECEIVED: 10318 messages - 1338737 bytes; 36369 LOOPS - 36.680 seconds - 36.50KB/s
[   micrortps_agent   ] SENT:     369 messages - 19188 bytes

321thijs123, 2022-05-12:
Hi, I'm running into the exact same issue, were you able to solve it?

321thijs123, 2022-05-12:
The cause of this problem for me was that ROS_DOMAIN_ID needs to be set to 0