PX4/px4_ros_com #122: Micro-RTPS Agent does not recieve any data from Micro-RTPS Client

k1r4n123, 2021-12-01:
I am using the CUAV V5+ flight controller and running PX4 version v1.12.3 and px4_fmu-v5_rtps on it. I have connected the flight controller to my Ubuntu 20.04 laptop via the USB port and by the DSU7 debugging port using a serial-USB converter. I access the debugging port using the following command: `screen /dev/ttyUSB0 57600 8N1`.

Within the debugging port I start the MicroRTPS client as so: 

[image]

And as you can see from the screenshot, I have ensured to stop all mavlink instances.

I start the MicroRTPS agent as so: 

[image]

I am also running the Sensor Combined Listener and Vehicle GPS Position Listener available in the px4_ros_com package, as so: 

[image]

[image]

I am expecting to see data being printed out in the Sensor Combined Listener and Vehicle GPS Position Listener applications, however I am not seeing anything. I am also not getting any errors or issues in the Micro RTPS Client or Agent. 

GPS data and Sensor data are being outputted from the flight controller as confirmed in the screenshot below:

[image]

... And after doing `sensors status`

[image]

So I am not sure why I am not seeing anything in Sensor Combined Listener and Vehicle GPS Position Listener. What am I doing wrong?

k1r4n123, 2021-12-02:
So now I have managed for the Agent to receive data however is there any way I can see what exactly are the messages it is receiving? Because it is receiving messages however, the Sensor Combined Listener and Vehicle GPS Position Listener applications are not receiving anything
[image]
. 
The way I did this was: 

- Run the micrortps client like : micrortps_client start /dev/ttyACM0 -b 115200
- Open another terminal and run: cat /dev/ttyACM0
- Run the micrortps agent like: micrortps_agent -d /dev/ttyACM0 -b 115200 -v
- Wait around 10 minutes for received messages to be accumulated in the agent

singhlok, 2022-03-06:
Hi,

I am facing the same issue. I can see that micrortps_agent is sending and receiving messages from micrortps_client.

However, Sensor Combined Listener is not displaying any.

Has this issue been resolved?

Thank you

senthilskm983, 2022-03-06:
https://discuss.px4.io/t/no-sensor-sensor-data-received-using-micrortps-client-agent-and-ros2/18756/4?u=sen.skm
Hope this will help.

singhlok, 2022-03-07:
> https://discuss.px4.io/t/no-sensor-sensor-data-received-using-micrortps-client-agent-and-ros2/18756/4?u=sen.skm
> Hope this will help.

Thank you Senthil. I will give it a try later today