PX4/px4_ros_com #123: build_ros2_workspace.bash failed

fd3s1, 2021-12-26:
./build_ros2_workspace.bash --no_ros1_bridge --ros_distro foxy --ros_path ~/ros2_foxy/ros2-linux/setup.bash
openjdk version "13.0.7" 2021-04-20
OpenJDK Runtime Environment (build 13.0.7+5-Ubuntu-0ubuntu120.04)
OpenJDK 64-Bit Server VM (build 13.0.7+5-Ubuntu-0ubuntu120.04, mixed mode, sharing)
[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[/opt/rti.com/rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.
[0.365s] WARNING:colcon.colcon_core.package_selection:ignoring unknown package 'ros1_bridge' in --packages-skip
Starting >>> px4_msgs
Finished <<< px4_msgs [9.94s]                     
Starting >>> px4_ros_com
--- stderr: px4_ros_com                               
c++: error: /home/sun2/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp: 没有那个文件或目录
c++: fatal error: no input files
compilation terminated.
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:497：CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_agent.cpp.o] 错误 1
make[2]: *** 正在等待未完成的任务....
make[1]: *** [CMakeFiles/Makefile2:142：CMakeFiles/micrortps_agent.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
make: *** [Makefile:141：all] 错误 2
---
Failed   <<< px4_ros_com [12.9s, exited with code 2]

Summary: 1 package finished [22.9s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com

I,m using ros foxy in ubuntu20.04 
I 'm struggling on it 
when I run  ./build_ros2_workspace.bash --no_ros1_bridge --ros_distro foxy --ros_path ~/ros2_foxy/ros2-linux/setup.bash in terminal in the ~/px4_ros_com_ros2/src/px4_ros_com/scripts path, this mistake will happen. I'm sure I have the correct version of fast DDS(the default version in ros-foxy) and fastrtpsgen(1.0.4), but may JDK version is 13.0.7, is this the problem?

fd3s1, 2021-12-26:
./build_ros2_workspace.bash --no_ros1_bridge --ros_distro foxy --ros_path ~/ros2_foxy/ros2-linux/setup.bash
openjdk version "13.0.7" 2021-04-20
OpenJDK Runtime Environment (build 13.0.7+5-Ubuntu-0ubuntu120.04)
OpenJDK 64-Bit Server VM (build 13.0.7+5-Ubuntu-0ubuntu120.04, mixed mode, sharing)
[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[/opt/rti.com/rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.
[0.365s] WARNING:colcon.colcon_core.package_selection:ignoring unknown package 'ros1_bridge' in --packages-skip
Starting >>> px4_msgs
Finished <<< px4_msgs [9.94s]                     
Starting >>> px4_ros_com
--- stderr: px4_ros_com                               
c++: error: /home/sun2/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp: 没有那个文件或目录
c++: fatal error: no input files
compilation terminated.
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:497：CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_agent.cpp.o] 错误 1
make[2]: *** 正在等待未完成的任务....
make[1]: *** [CMakeFiles/Makefile2:142：CMakeFiles/micrortps_agent.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
make: *** [Makefile:141：all] 错误 2
---
Failed   <<< px4_ros_com [12.9s, exited with code 2]

Summary: 1 package finished [22.9s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com

Li-Jinjie, 2022-01-13:
> c++: error: /home/sun2/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp: 没有那个文件或目录 (no such directory)

This error happens because fastrtpsgen does not work normally. For me, when I `cd /path_to_ros2_ws/build/px4_ros_com` and execute `make`, the console outputs

_Generating micro-RTPS agent code...
Failed to import genmsg: No module named 'genmsg'_

So just install python3-genmsg by inputing 
`sudo apt-get install python3-genmsg`
This works for me.

Besides, I upload my docker image to docker hub: https://hub.docker.com/r/lijinjie/px4-ros2-arm64 , which is verified on TX2 NX.

I hope this method can solve your problem :)