PX4/px4_ros_com #124: Added topics not visible in ROS 2

luisp23, 2021-12-28:
Hello, I am trying to add new topics to the urtps_bridge_topics.yaml. I followed the same detailed steps [here](https://docs.px4.io/master/en/ros/ros2_offboard_control.html). I've installed and used everything [here](https://docs.px4.io/master/en/ros/ros2_comm.html)

I add the following to `PX4-Autopilot/msg/tools/urtps_bridge_topics.yaml`

```yaml
 ...
 - msg:     sensor_combined
    send:    true
  - msg:     vehicle_local_position
    send:    true
  - msg:     vehicle_imu
    send:    true
```

I then use the uorb_to_ros_urtps_topics.py to generate this file in `px4_ros_com`

```bash 
cd ~/PX4-Autopilot/msg/tools
python3 uorb_to_ros_urtps_topics.py -i urtps_bridge_topics.yaml -o ~/px4_ros_com_ros2/src/px4_ros_com/templates/urtps_bridge_topics.yaml
```

This succesfully overwrites the file and appends the following to `~/px4_ros_com_ros2/src/px4_ros_com/templates/urtps_bridge_topics.yaml`: 

```yaml
 ...
  - msg: SensorCombined
    send: true
  - msg: VehicleLocalPosition
    send: true
  - msg: VehicleImu
    send: true
```

I've made sure to clean all builds: 
```bash 
cd ~/PX4-Autopilot
make clean 
rm -rf build/
```
```bash 
cd ~/px4_ros_com_ros2/src/px4_ros_com/scripts 
source clean_all.bash 
```

I then build `px4_ros_com`:
```bash 
cd ~/px4_ros_com_ros2/src/px4_ros_com/scripts 
source build_ros2_workspace.bash 
```
This successfully builds. I then build the px4_sitl_rtps: 
```bash 
make px4_sitl_rtps gazebo
```

This build successfully and I am able to see the iris in Gazebo.

I have attached  the output of this here:
[make_px4_sitl_rtps_gazebo_output.txt](https://github.com/PX4/px4_ros_com/files/7782225/make_px4_sitl_rtps_gazebo_output.txt)

I then run the `micrortps_agent` and get the following output:

```bash 

micrortps_agent start -t UDP

--- MicroRTPS Agent ---
[   micrortps_agent   ]	Starting link...
[   micrortps_agent   ]	UDP transport: ip address: 127.0.0.1; recv port: 2020; send port: 2019
[ micrortps_transport ]	UDP transport: Trying to connect...
[ micrortps_transport ]	UDP transport: Connected to server!

---   Subscribers   ---
- DebugArray subscriber started
- DebugKeyValue subscriber started
- DebugValue subscriber started
- DebugVect subscriber started
- OffboardControlMode subscriber started
- OpticalFlow subscriber started
- PositionSetpoint subscriber started
- PositionSetpointTriplet subscriber started
- TelemetryStatus subscriber started
- Timesync subscriber started
- VehicleCommand subscriber started
- VehicleLocalPositionSetpoint subscriber started
- VehicleTrajectoryWaypoint subscriber started
- OnboardComputerStatus subscriber started
- TrajectoryBezier subscriber started
- VehicleTrajectoryBezier subscriber started
- TrajectorySetpoint subscriber started
- VehicleMocapOdometry subscriber started
- VehicleVisualOdometry subscriber started
-----------------------

----   Publishers  ----
- Timesync publishers started
- TrajectoryWaypoint publisher started
- VehicleControlMode publisher started
- VehicleOdometry publisher started
- VehicleStatus publisher started
- CollisionConstraints publisher started
- TimesyncStatus publisher started
- SensorCombined publisher started
- VehicleLocalPosition publisher started
- VehicleImu publisher started
- VehicleTrajectoryWaypointDesired publisher started
-----------------------

```
However, when I list the available ros2 topics I get the following: 

```bash 
ros2 topic list
/fmu/collision_constraints/out
/fmu/debug_array/in
/fmu/debug_key_value/in
/fmu/debug_value/in
/fmu/debug_vect/in
/fmu/offboard_control_mode/in
/fmu/onboard_computer_status/in
/fmu/optical_flow/in
/fmu/position_setpoint/in
/fmu/position_setpoint_triplet/in
/fmu/telemetry_status/in
/fmu/timesync/in
/fmu/timesync/out
/fmu/trajectory_bezier/in
/fmu/trajectory_setpoint/in
/fmu/trajectory_waypoint/out
/fmu/vehicle_command/in
/fmu/vehicle_control_mode/out
/fmu/vehicle_local_position_setpoint/in
/fmu/vehicle_mocap_odometry/in
/fmu/vehicle_odometry/out
/fmu/vehicle_status/out
/fmu/vehicle_trajectory_bezier/in
/fmu/vehicle_trajectory_waypoint/in
/fmu/vehicle_visual_odometry/in
/parameter_events
/rosout
/timesync_status
```

I do not see the `vehicle_local_position` or `vehicle_imu` topics published which I am assuming would be listed as `/fmu/vehicle_local_position/out` and `/fmu/vehicle_imu/out`. 

Am I doing any of these steps in the wrong order or is there a step that I am missing? Any help would be greatly appreciated thank you!

luisp23, 2022-01-03:
After going through these steps, something seems to get messed up with my ROS2 system in the background. I can see the the new topics only after restarting the ROS2 daemon with the following: 

```bash 
ros2 daemon stop 
ros2 daemon start
```
This only seems to be required once after going through the above steps.

closed 2022-01-03 by luisp23

JohnnyLiu2886, 2022-01-19:
> After going through these steps, something seems to get messed up with my ROS2 system in the background. I can see the the new topics only after restarting the ROS2 daemon with the following:
> 
> ```shell
> ros2 daemon stop 
> ros2 daemon start
> ```
> 
> This only seems to be required once after going through the above steps.

Did you subscribe to vehicle_local_position successfully?
I can subscribe to any uorb message added by me in the .yaml file.
Only vehicle_local_position is not working.

I can found /fmu/vehicle_local_position/out using "ros2 topic list" and found it matched in the micrortps_agent terminal. It is also possible to listen to its data using "listener vehicle_local_position".

However, I am still unable to subscribe to the "vehicle_local_position" topic using my ros2 code and even "ros2 topic echo /fmu/vehicle_local_position".

1. try it with the latest commit of PX4-Autopilot.
2. try it with "estimator_local_position" and "vehicle_local_position_groundtruth" which are based on "vehicle_local_position"
3. try it outside the office in order to receive the gps position of the reference position. Check the terminal using ssh with my laptop

However, all of them are failed. I cannot understand why it is not publishing even after reading the source code of ekf2_main.cpp. 

Here is my situation. I have tried a lot of method:
[image]

JohnnyLiu2886, 2022-01-20:
It is working on the gazebo simulator now but it is still not working in the holybro pixhawk 4.