PX4/px4_ros_com #125: will ros2 launch mavros?

Boshen97, 2021-12-30:
Hello, I have a question to consult.

Are there any plans to introduce an abstraction layer similar to the mavros function package in ROS2?

TSC21, 2021-12-30:
@Boshen97 yes. It is in our plans to provide an API to facilitate the usage of the bridge. More info can be found on the presentations this year on the PX4 Dev Summit and ROS World.

TSC21, 2021-12-30:
@Boshen97 also to clarify: we will have an abstraction layer that interfaces with the microRTPS bridge, not with MAVROS. px4_ros_com is completely different approach to bridge PX4 and ROS2. So nothing here interfaces or launches MAVROS.

Boshen97, 2021-12-31:
Thanks for your reply, now I understand

Boshen97, 2021-12-31:
When is the launch time of this abstraction layer？

TSC21, 2021-12-31:
> When is the launch time of this abstraction layer？

No current timeline. Something to aim on the next 2/3 months.