PX4/px4_ros_com #127: how to control vehicle to takeoff in offboard via VehicleCommand::VEHICLE_CMD_NAV_TAKEOFF?

xiaobai-cai-hold, 2022-01-14:
I find that someone control vehicle to takeoff in offboardcontrol.cpp by using (publish_offboard_control_mode()  publish_trajectory_setpoint(0.0, 0.0, 2) ).

however , I found VehicleCommand::VEHICLE_CMD_NAV_TAKEOFF  in vehiclecommand.msg, I am confused ,why not use it to control vehicle to takeoff?

I  test this command by using the following code , I failed in gazebo, how to use VehicleCommand::VEHICLE_CMD_NAV_TAKEOFF to takeoff?

      void OffboardControl::publish_takeoff_from_local_position() const
      {
	      VehicleCommand msg{};
	      msg.timestamp = timestamp_.load();
	      msg.param1 = 0.0;//pitch
	      msg.param2 = NAN;
	      msg.param3 = NAN;
	      msg.param4 = NAN;//yaw
	      msg.param5 = this->lat;
	      msg.param6 = this->lon;
	      msg.param7 = this->alt +20;
	      msg.command = VehicleCommand::VEHICLE_CMD_NAV_TAKEOFF;
	      msg.target_system = 1;
	      msg.target_component = 1;
	      msg.source_system = 1;
	      msg.source_component = 1;
	      msg.from_external = true;
      
	      vehicle_command_publisher_->publish(msg);
      
      }

I am a student ,first to touch this ,can you provide some instruction to help me learn to control vehicle via ros2? thanks

xiaobai-cai-hold, 2022-01-14:
At first I make the vehicle to work offboard mode ,  I send 

publish_vehicle_command(VehicleCommand::VEHICLE_CMD_NAV_VTOL_LAND) or publish_vehicle_command(VehicleCommand::VEHICLE_CMD_DO_SET_MODE,1.0, 4.0,2.0)

the vehicle in gazebo can land successfully.

According to the way I send this command 

publish_vehicle_command(VehicleCommand::VEHICLE_CMD_NAV_VTOL_TAKEOFF) 

however it dosen't takeoff ,  a little upset.

TheSpecialOne45, 2023-04-29:
Hi @xiaobai-cai-hold !

I'm reviving this thread as I have the same problem, were you able to solve it in the end (taking off using ROS2, without using the offboard position mode) ?

AumJasani, 2024-05-03:
I tried to do the same thing but I haven't found any success. I found that if the vehicle is in `OFFBOARD MODE`, the `VEHICLE_CMD_NAV_TAKEOFF` won't work. But, even if it is not on offboard mode, I am getting this message `WARN [navigator] Already higher than takeoff altitude` and the vehicle disarms it self after spinning the props for 1 or 2 secs.

fnndyl, 2024-10-30:
Throwing in my two sense that I have the same issue as @AumJasani, very odd