PX4/px4_ros_com #128: micrortps_agent logging with rclcpp instead of printf and cout

benjinne, 2022-01-15:
Utilizing ros logging will allow the easy ability to control when logging is enabled or disabled, also with other benefits.
https://docs.ros.org/en/galactic/Concepts/About-Logging.html

I'm currently getting too many messages about publishers and subscribers matching which is essentially spam.
https://github.com/PX4/px4_ros_com/blob/6f182bc19477b942d7ca35251fbf3e0a08eb72de/templates/Publisher.cpp.em#L207
```
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleStatus publisher matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleLandDetected publisher matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleStatus publisher matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleOdometry publisher matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	OffboardTrajectorySetpoint subscriber matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleAttitudeSetpoint subscriber matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	OffboardControlMode subscriber matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	OffboardControlMode subscriber matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	Timesync publisher matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleStatus publisher matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleLandDetected publisher matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	SensorCombined publisher matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleGpsPosition publisher matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	SensorCombined publisher unmatched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleStatus publisher matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleStatus publisher matched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleStatus publisher unmatched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleStatus publisher unmatched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleLandDetected publisher unmatched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleStatus publisher unmatched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleStatus publisher unmatched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleGpsPosition publisher unmatched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleOdometry publisher unmatched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleCommand subscriber unmatched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	VehicleStatus publisher unmatched[0m
[micrortps_agent-5] [0;37m[   micrortps_agent   ]	Timesync publisher unmatched[0m
```

TSC21, 2022-01-18:
@benjinne you can always disable it by removing it from the templates. The above only happens when you stop publishing/subscribing. I would only see it as spam if these were recurrent, paced logging messages. Closing.

closed 2022-01-18 by TSC21

benjinne, 2022-01-18:
Hey @TSC21 I opened this more as an idea, not a bug. I've been looking into how I can add ros2 logging so that messages can be filtered by a log level. The matching and un-matching could be debug for example

benjinne, 2022-01-19:
So I tried, using RCLCPP_DEBUG in the templates like this:
[image]
And it appears to work, but when I try launching with `use_sim_time` I get
```
micrortps_agent -t UDP --ros-args -p use_sim_time:=true --log-level micrortps_agent:=WARN
--- MicroRTPS Agent ---
micrortps_agent: unrecognized option '--log-level'
[ERROR]: Failed to parse global arguments
terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
  what():  failed to initialize rcl: Couldn't parse parameter override rule: '-p --log-level'. Error: Expecting token or wildcard, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/arguments.c:1170, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/arguments.c:343
Aborted (core dumped)
```
Reversing the arguments gets a similar error
```
micrortps_agent -t UDP --ros-args --log-level micrortps_agent:=WARN -p use_sim_time:=true
--- MicroRTPS Agent ---
micrortps_agent: invalid option -- 'p'
[ERROR]: Failed to parse global arguments
terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
  what():  failed to initialize rcl: Couldn't parse log level: '--log-level -p'. Error: Argument does not use a valid severity level, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/arguments.c:1767, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/arguments.c:432
Aborted (core dumped)
```

I've tried several ways of organizing the parameters, but nothing appears to work. I even added the executable to the lib folder so I could use `ros2 run` to run it, but I still get the same problem. I compared this to running other ros2 executables and I'm able to pass the same arguments with no problems.

TSC21, 2022-01-19:
The microRTPS agent is not a ROS2 node, so you can't pass the arguments the same way.

TSC21, 2022-01-19:
> I understand, however using `micrortps_agent -t UDP --ros-args --log-level micrortps_agent:=INFO` appears to work since there's only one ros argument. Also `--ros-args use_sim_time:=true` is a feature that also works.

Correct and I didn't say it is not supposed to work. You just have to take into account that the params are not parsed the same way as this was a ROS2 node.

TSC21, 2022-01-19:
> the microRTPS agent also uses `rclcpp::init(argc, argv)` just like any other ros executable and it should work, but there's something strange going on. A node does not need to be created to get the ros2 logging functionality.

The microRTPS agent only uses the rclcpp init when using sim time. This is not applicable by default and it's not a standard functionality because, again, the agent should not be interpreted or used as a ROS2 node.

benjinne, 2022-01-19:
Ah okay, thanks for clarifying. I'm just going to take the simple approach and comment out the lines I don't want publishing for now. I might revisit this later