PX4/px4_ros_com #131: bash build_ros2_workspace.bash failed

REGATTE, 2022-02-17:
Hi,
I'm building the PX4-ROS2 Communication system using the FastDDS implementation as of the doc's in PX4 website.

Ive installed all the required libraries, dependencies and made sure the versions are correct.
Im running ROS2 Foxy and no ros1-2Bridge so ran the bash script with the following arguments:

cd /to/px4_ros_com/scripts
```bash build_ros2_workspace.bash --no_ros1_bridge --ros_distro foxy --ros_path ~/ros2_foxy/install/setup.bash```

and i get the following errors:

```
Starting >>> px4_msgs
Finished <<< px4_msgs [24.1s]                       
Starting >>> px4_ros_com
--- stderr: px4_ros_com                              
c++: error: /home/ashok/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp: No such file or directory
c++: fatal error: no input files
compilation terminated.
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:497: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_agent.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< px4_ros_com [1.97s, exited with code 2]
```

Can someone help me on how to get this sorted out.

Thank you.

REGATTE, 2022-02-18:
the error got fixed, once I set my JAVA JDK to 11. It was at 13, due to the Ubuntu.sh from PX4-Autopilot/Tools.

closed 2022-02-18 by REGATTE

mengchaoheng, 2022-10-16:
> the error got fixed, once I set my JAVA JDK to 11. It was at 13, due to the Ubuntu.sh from PX4-Autopilot/Tools.

@REGATTE how to set Java idk to 11? I am a new person on that

REGATTE, 2022-10-24:
Use SDKMAN and Gradle, then install Hava JDK from the official Oracle website.