PX4/px4_ros_com #135: Error while trying to read serial port configuration on Raspberry Pi

TheLonelyFighter, 2022-04-05:
I run the micrortps_agent on a Raspberry Pi and I get the following error when starting the agent:

`Error while trying to read serial port configuration`

The agent runs fine on my laptop, but not on the Raspberry Pi. I suggest modifying the microRTPS_transport.cpp code to check whether the device it runs on can use the function `ioctl(_uart_fd, TIOCGSERIAL, &serial_ctl) ` without errors.

To fix the error, I ended up commenting this entire section from microRTPS_transport.cpp and then rebuild the px4_ros_com package:
```
#if defined(__linux__) || defined(__PX4_LINUX)
	// For Linux, set high speed polling at the chip level. Since this routine relies on a USB latency
	// change at the chip level it may fail on certain chip sets if their driver does not support this
	// configuration request
	{
		struct serial_struct serial_ctl;

		if (ioctl(_uart_fd, TIOCGSERIAL, &serial_ctl) < 0) {
			printf("\033[0;31m[ micrortps_transport ]\tError while trying to read serial port configuration: %d\033[0m\n", errno);

			if (ioctl(_uart_fd, TCFLSH, TCIOFLUSH) == -1) {
				int errno_bkp = errno;
				printf("\033[0;31m[ protocol__splitter ]\tCould not flush terminal\033[0m\n");
				close();
				return -errno_bkp;
			}
		}

		serial_ctl.flags |= ASYNC_LOW_LATENCY;

		if (ioctl(_uart_fd, TIOCSSERIAL, &serial_ctl) < 0) {
			int errno_bkp = errno;
			printf("\033[0;31m[ micrortps_transport ]\tError while trying to write serial port latency: %d\033[0m\n", errno);
			close();
			return -errno_bkp;
		}
	}
#endif /* __linux__ */
```

gustavotbarros, 2023-02-15:
@TheLonelyFighter Thanks! I had this same problem for more than a week. I did the same thing and it worked perfectly.

TheLonelyFighter, 2023-02-16:
Great, I'm glad I could help ;)