PX4/px4_ros_com #148: Colcon build stuck at 93%

EdmarCaixeta, 2022-08-08:
Hi,

I'm currently using Ubuntu 22.04 and everything works fine. I can run "make px4_sitl_rtps gazebo" and then my computer builds px4 and opens gazebo. But when I try to build the example ROS2 workspace, my computer builds the px4_msgs completely but then when building px4_ros_com it stucks at 93%. I tried to build manually by running "colcon build" at the root of the workspace with and without the --symlink-install and tried to build adapting the "build_ros2_workspace.sh" script to Ubuntu 22.04 (Just changed the OS verifications and set the ROS version to Humble). I tried to change my java version (11 to 8), checked my FastDDS and FastDDSGen versions and everything seems fine. 

What should I do? 

PS: Sorry if my informations its not good enough it's my first Issue :)

EdmarCaixeta, 2022-08-08:
Maybe it is an useless information but Im using fish shell but the error persists on bash as well

erisevve, 2022-08-12:
I was having similar issues on Ubuntu 22.04 and humble (built inside of docker). I tried building a workspace with only the PX4-Autopilot, px4_msgs and px4_ros_com packages and the build didn't progress beyond 95 %, stopping after the micrortps_agent target hade been 'finished': `[95%] Built target micrortps_agent`. I compared the output to the same install and build process in a dockerfile with galactic and Ubuntu 20.04, where the next step was `[95%] Built target frame_transforms`. Guessing that _frame_transforms_ was the culprit I removed it from the cmake-file, wherupon the humble version finished the colcon build and worked. So I suppose the issue is in the frame_transforms executable. Very odd that there is no error and simply an infinite build process. 

Beyond the scripts updated in the latest commit I noticed that the script _generate_microRTPS_bridge.py_ seems to have only been updated in the PX4-Autopilot Tools folder and not inside of the scripts folder in this repository. I'm not sure which script is used while building, but I modified the one inside of this repository  to match the one in the PX4-autopilot.  I had done this already when I made the above change to the cmake-file, so I'm not sure whether or not it is necessary.  

TLDR;  colcon finishes the micrortps_agent target inside of the cmake, but does not progress to the installation step. Removing frame_transforms from build (and possibly updating  _generate_microRTPS_bridge.py_) worked as a quick fix.

WarriorHanamy, 2022-08-23:
I was having similar issues on Ubuntu 22.04 and humble(vmware), I was stuck at 100%.
First, when source build_ros2_workspace.bash in (px4_ros_com_ros2/src/px4_ros_com/script), I got terminal crash problem which caused by when source any ***.bash, the command in ***.bash $0 will give the name of bash, that is /bin/bash, other than the name of the file itself (***.bash). So I reduced to
bash build_ros2_workspace.bash. （I tested in my friend's computer, and There is no problem in virtual-machine-free enviroment).
Then, I got the simililar issue stuck in build process 100%, I just waited to finish, and it took 30 minutes to finish (Maybe)

PR #158 (2022-11-03): Fix frame_transform link issue
Fixes #148
closed 2022-11-03 by TSC21