PX4/px4_ros_com #152: Bad CRC Errors ; Unexpected byte value in Cdr::deserialize(bool),expected 0 or 1

Aravind-Adhith, 2022-08-25:
Hi @TSC21 @dagar ,

We am trying to establish the px4 ros2 bridge using serial communication. We tried the following versions but RTPS build for the fmu-v5 fails. 

Can you please point us to the right repos. 
Ubuntu LTS 20.04
px4_ros_com - master
px4_msgs - master
Fast RTPS version - 1.0.4 
PX4 Autopilot - master
For this above combination, we are not able to build the RTPS firmware for the Pixhawk 4 and 6C. 

Ubuntu LTS 20.04
px4_ros_com - master
px4_msgs - master
Fast RTPS version - 1.0.4 
PX4 Autopilot - v1.13.0
For this combination, the colcon build for the px4_ros_com fails, even when we generate the px4_msgs from the "PX4-Autopilot/blob/main/msg/tools/uorb_to_ros_urtps_topics.py". 

Any help for this is appriciated. 

[image]

[image]

Aravind-Adhith, 2022-08-26:
Update :
The below configuration allows us to run the agent, but we are getting **Bad CRC errors**. And it still crashes when we set the Baud rate greater than or equal to 115200.

Ubuntu LTS 20.04
px4_ros_com - April 2022 commit ; SHA1 : 1562ff30d56b7ba26e4d2436724490f900cc2375
px4_msgs - master - but msgs generated from Firmware v1.13.0 and copied
Fast RTPS version - 1.0.4
PX4 Autopilot - v1.13.0

mengchaoheng, 2022-08-30:
the same problem
```Console
parallels@ubuntu-linux-20-04-desktop:~/px4_ros_com_ros2/src/px4_ros_com$ micrortps_agent -t UDP
--- MicroRTPS Agent ---
[   micrortps_agent   ]	Starting link...
[   micrortps_agent   ]	UDP transport: ip address: 127.0.0.1; recv port: 2020; send port: 2019
[ micrortps_transport ]	UDP transport: Trying to connect...
[ micrortps_transport ]	UDP transport: Connected to server!

---   Subscribers   ---
- DebugArray subscriber started
- DebugKeyValue subscriber started
- DebugValue subscriber started
- DebugVect subscriber started
- OffboardControlMode subscriber started
- SensorOpticalFlow subscriber started
- PositionSetpoint subscriber started
- PositionSetpointTriplet subscriber started
- TelemetryStatus subscriber started
- Timesync subscriber started
- VehicleCommand subscriber started
- VehicleLocalPositionSetpoint subscriber started
- TrajectorySetpoint subscriber started
- VehicleTrajectoryWaypoint subscriber started
- OnboardComputerStatus subscriber started
- TrajectoryBezier subscriber started
- VehicleTrajectoryBezier subscriber started
- VehicleMocapOdometry subscriber started
- VehicleVisualOdometry subscriber started
-----------------------

----   Publishers  ----
- Timesync publishers started
- TrajectoryWaypoint publisher started
- VehicleControlMode publisher started
- VehicleOdometry publisher started
- VehicleStatus publisher started
- CollisionConstraints publisher started
- TimesyncStatus publisher started
- SensorCombined publisher started
- VehicleTrajectoryWaypointDesired publisher started
-----------------------
terminate called after throwing an instance of 'eprosima::fastcdr::exception::BadParamException'
  what():  Unexpected byte value in Cdr::deserialize(bool), expected 0 or 1
Aborted (core dumped)

```

vegaf1, 2022-09-12:
Same error as @mengchaoheng

maxpolzin, 2022-09-16:
Did you try regenerating the ROS message definitions from your PX4-Autopilot repo using the respective script, see https://discuss.px4.io/t/issues-running-ros2-px4/28400/20?u=maxinho