PX4/px4_ros_com #161: Any alternative to Mavros udp broadcast to QGC in ROS2?

shazib2t, 2022-10-07:
Need some suggestion: I have been using ros1 and mavros for my companys multi uav and companion computer setup with 2.4ghz connection to GCS computer. Until now I have been successfully using the mavros with its udp broadcaster to the GCS computer for using QGC. But now if we have to move to ros2 and px4-ros-com what is my option as opposed to udp broadcaster to QGC. This is the only way I can communicate using QGC as of now.

shazib2t, 2022-11-02:
@ripdk12 I asked the same question on the PX4 discord channel and the response is, [Answer here](https://discord.com/channels/1022170275984457759/1030186476153085993/1031600738176090223). Basically, the PX4 team wants us to use PX4_ROS_COM with MicroROS XRCE DDS. With this method, we have to use the MAVLINK router for GCS.

dagar, 2022-11-02:
Mavlink router should work well for that in the meantime.

It's going to take a bit to get there, but ultimately I'd like to be able to simply run a PX4 mavlink (module) instance directly in ROS2 as a node.

Kihoon-Shin, 2023-03-24:
So, You are using Mavlink_router ?
Is it working with ROS2?