PX4/px4_ros_com #166: Is it supposed to generate micrortps_agent and micro-ros-agent?

ElliotHYLee, 2022-11-09:
I'm trying to setup PX4 and ROS2 communication. All builds are successful with the source /opt/ros/~~ and source /install/setup.bash

However, the bash shell cannot find micrortps_agent and cannot run the listener.

I have been following: https://docs.px4.io/main/en/ros/ros2_comm.html#install-fast-dds

### **1. docker, ubunt20, ros2 foxy, (FROM osrf/ros:foxy-desktop)  - check**

echo $ROS_DOMAIN_ID --> nth comes out (checking unset)

### **2.PX4 environment setup - check (being able to build and upload)**
PX4_Autopilot - branch : release/1.13
make px4_sitl_rtps gazebo

### **3. on U20 with ros2 foxy - with fast rtps - check**

dpkg -l | grep fastrt
ii  ros-foxy-fastrtps                               2.1.2-1focal.20220829.174844      amd64        Implementation of RTPS standard.
ii  ros-foxy-fastrtps-cmake-module                  1.0.4-1focal.20220829.181444      amd64        Provide CMake module to find eProsima FastRTPS.
ii  ros-foxy-rmw-fastrtps-cpp                       1.3.1-1focal.20221012.224708      amd64        Implement the ROS middleware interface using eProsima FastRTPS static code generation in C++.
ii  ros-foxy-rmw-fastrtps-shared-cpp                1.3.1-1focal.20221012.221742      amd64        Code shared on static and dynamic type support of rmw_fastrtps_cpp.
ii  ros-foxy-rosidl-typesupport-fastrtps-c          1.0.4-1focal.20221012.215818      amd64        Generate the C interfaces for eProsima FastRTPS.
ii  ros-foxy-rosidl-typesupport-fastrtps-cpp        1.0.4-1focal.20221012.215633      amd64        Generate the C++ interfaces for eProsima FastRTPS.

fastrtpsgen -version

openjdk version "11.0.17" 2022-10-18
OpenJDK Runtime Environment (build 11.0.17+8-post-Ubuntu-1ubuntu220.04)
OpenJDK 64-Bit Server VM (build 11.0.17+8-post-Ubuntu-1ubuntu220.04, mixed mode, sharing)
fastrtpsgen version 1.0.4

### **4. git cloned px4_msgs and px4_ros_com under ros workspace/src and run build - check**

root@elcom:/home/ubuntu/projects/robot_ws/src/px4_ros_com/scripts# pwd
/home/ubuntu/projects/robot_ws/src/px4_ros_com/scripts
root@elcom:/home/ubuntu/projects/robot_ws/src/px4_ros_com/scripts# bash build_ros2_workspace.bash 
Starting >>> micro_ros_msgs
[0.378s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/ubuntu/projects/robot_ws/install/micro_ros_setup' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.380s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/ubuntu/projects/robot_ws/install/micro_ros_setup' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> px4_msgs
Starting >>> drive_base_msgs
Finished <<< micro_ros_msgs [15.1s]                                            
Starting >>> micro_ros_agent
Finished <<< drive_base_msgs [16.0s]                               
[Processing: micro_ros_agent, px4_msgs]                            
Finished <<< micro_ros_agent [35.8s]                               
[Processing: px4_msgs]                                        
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                        
Finished <<< px4_msgs [2min 48s]                           
Starting >>> px4_ros_com
Finished <<< px4_ros_com [14.1s]                           

Summary: 5 packages finished [3min 2s]

ROS2 workspace ready...

### **5. micrortps_agent -t UDP - error**
bash: /home/ubuntu/projects/robot_ws/install/micrortps_agent/bin/micrortps_agent: No such file or directory

**5.a My Improvisation**
cd robot_ws/src/
git clone https://github.com/PX4/micrortps_agent
cd robot_ws/
colon build
source install/setup.bash

**micrortps_agent -t UDP - error****
--- MicroRTPS Agent ---
[   micrortps_agent   ] Starting link...
[   micrortps_agent   ] UDP transport: ip address: 127.0.0.1; recv port: 2020; send port: 2019
[ micrortps_transport ] UDP transport: Trying to connect...
[ micrortps_transport ] UDP transport: Connected to server!

---   Subscribers   ---
- debug_array subscriber started
- debug_key_value subscriber started
- debug_value subscriber started
- debug_vect subscriber started
- offboard_control_mode subscriber started
- sensor_optical_flow subscriber started
- position_setpoint subscriber started
- position_setpoint_triplet subscriber started
- telemetry_status subscriber started
- timesync subscriber started
- vehicle_command subscriber started
- vehicle_local_position_setpoint subscriber started
- vehicle_attitude_setpoint subscriber started
- vehicle_rates_setpoint subscriber started
- trajectory_setpoint subscriber started
- vehicle_trajectory_waypoint subscriber started
- onboard_computer_status subscriber started
- trajectory_bezier subscriber started
- vehicle_trajectory_bezier subscriber started
- vehicle_mocap_odometry subscriber started
- vehicle_visual_odometry subscriber started
-----------------------

----   Publishers  ----
- timesync publishers started
- trajectory_waypoint publisher started
- vehicle_control_mode publisher started
- vehicle_odometry publisher started
- vehicle_status publisher started
- collision_constraints publisher started
- timesync_status publisher started
- sensor_combined publisher started
- vehicle_trajectory_waypoint_desired publisher started
-----------------------
[   micrortps_agent   ] Unexpected topic ID '16' to publish. Please make sure the agent is capable of parsing the message associated to the topic ID '16'
[   micrortps_agent   ] Unexpected topic ID '27' to publish. Please make sure the agent is capable of parsing the message associated to the topic ID '27'
[   micrortps_agent   ] Unexpected topic ID '16' to publish. Please make sure the agent is capable of parsing the message associated to the topic ID '16'
[   micrortps_agent   ] Unexpected topic ID '27' to publish. Please make sure the agent is capable of parsing the message associated to the topic ID '27'
[   micrortps_agent   ] Unexpected topic ID '27' to publish. Please make sure the agent is capable of parsing the message associated to the topic ID '27'

**### 5.b thoughs**
After adding micrortps_agent under /src/ and colcon build, the micrortps_agent works. Now, there's msg saying unexpected topic 16 and 27. It seems that the parser is not working properly with px4_msgs.

Based on the documentaion, I assumed it should have worked just by px4_ros_com.

### **6 ros2 launch px4_ros_com sensor_combined_listener.launch.py - error**

[INFO] [launch]: All log files can be found below /root/.ros/log/2022-11-09-20-58-49-644613-elcom-181454
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [micro_ros_agent udp4 --port 8888 -v -1]: process started with pid [181458]
[INFO] [sensor_combined_listener-2]: process started with pid [181460]
[micro_ros_agent udp4 --port 8888 -v -1] /bin/sh: 1: micro_ros_agent: not found
[ERROR] [micro_ros_agent udp4 --port 8888 -v -1]: process has died [pid 181458, exit code 127, cmd 'micro_ros_agent udp4 --port 8888 -v '].

**6.amy improvisation**
By following the documentation, it does not know the cmd, micro_ros_agent.

By, https://micro.ros.org/docs/tutorials/core/first_application_linux,

git clone https://github.com/micro-ROS/micro_ros_setup

colcon build and source install/setup.bash
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
source install/local_setup.bash

root@elcom:/home/ubuntu/projects/robot_ws# micro_ros_agent
bash: micro_ros_agent: command not found

**ros2 run micro_ros_agent** 
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: argument Call `ros2 <command> -h` for more detailed usage.: invalid choice: 'micro_ros_agent' (choose from 'action', 'bag', 'component', 'daemon', 'doctor', 'extension_points', 'extensions', 'interface', 'launch', 'lifecycle', 'multicast', 'node', 'param', 'pkg', 'run', 'security', 'service', 'topic', 'wtf')

ros2 run micro_ros_agent does not have option for UDP..

**Conclusion**
At my best understading, by following the doc, https://docs.px4.io/main/en/ros/ros2_comm.html#install-fast-dds, 

micrortps_agent: command not found
micro_ros_agent: command not found

I wonder if this repo is supposed to install these two commands.

If not, where can I get them? Any advise?

zxh0916, 2022-11-12:
Same problem on step 5.
> [ micrortps_agent ] Unexpected topic ID '16' to publish. Please make sure the agent is capable of parsing the message associated to the topic ID '16'
> [ micrortps_agent ] Unexpected topic ID '27' to [...]

zxh0916, 2022-11-12:
> thank you

Hi! I just solved the problem on step 5.
I'm using ubuntu 20.04 and ros2 **Galactic**.
I cloned PX4-Autopilot repo of version 1.13.1 by `git clone -b v1.13.1 https://github.com/PX4/PX4-Autopilot.git --recursive`
After cloned px4_ros_com and px4_msg, i switched branch by `cd ~/px4_ros_com_ros2/src/px4_msgs && git checkout daee121` and `cd ~/px4_ros_com_ros2/src/px4_ros_com && git checkout 7e25c34`
Then i used the python script inside the PX4-Autopilot repo to convert the messages and the message list to **overwrite** the messages and list in px4_ros_com and px4_msg:
`python ~/PX4-Autopilot/msg/tools/uorb_to_ros_msgs.py ~/PX4-Autopilot/msg/ ~/px4_ros_com_ros2/src/px4_msgs/msg/`
then
`python ~/PX4-Autopilot/msg/tools/uorb_to_ros_urtps_topics.py -i ~/PX4-Autopilot/msg/tools/urtps_bridge_topics.yaml -o ~/px4_ros_com_ros2/src/px4_ros_com/templates/urtps_bridge_topics.yaml`
And rebuild the workspace by running `cd ~/px4_ros_com_ros2/src/px4_ros_com/scripts/` and `./build_ros2_workspace.bash --verbose --no_ros1_bridge --ros_distro galactic --ros_path /opt/ros/galactic/setup.bash`.
But the build still fails because the definition of message **TrajectorySetpoint** dose not met the requirement of script `~/px4_ros_com_ros2/src/px4_ros_com/src/examples/offboard/offboard_control.cpp`.
So i modified the script around line 175 by comment `msg.position = {0.0, 0.0, -5.0};` and replace it by `msg.x = 0.0; msg.y = 0.0; msg.z = -5.0;` and rebuilt the workspace.
Then all commands ran properly.
forgive my poor english :)