PX4/px4_ros_com #167: The c++ offboard example is not support the latest microdds version.

JetJie, 2022-11-11:
IThe c++ offboard example is not support the latest microdds version.  Is there any update on the offboard control example?

dagar, 2022-11-11:
Take a look at https://github.com/Jaeyoung-Lim/px4-offboard.

At this point with microdds I'm not sure if we should keep px4_ros_com or just get it split up into separate standalone examples we can refer to.

laylow13, 2022-11-17:
I got QoS errors when running offboard_control, so I made some changes in offboard_control.cpp and it works fine now.
`rclcpp::QoS qos(10);
		qos.best_effort();
		qos.transient_local();

		offboard_control_mode_publisher_ = this->create_publisher<OffboardControlMode>("/fmu/in/offboard_control_mode", qos);
		trajectory_setpoint_publisher_ = this->create_publisher<TrajectorySetpoint>("/fmu/in/trajectory_setpoint", qos);
		vehicle_command_publisher_ = this->create_publisher<VehicleCommand>("/fmu/in/vehicle_command", qos);`

laylow13, 2022-11-17:
I got a QoS error when I was running offboard_control. After making the following changes in the code, it can run normally
```
                rclcpp::QoS qos(10);
		qos.best_effort();
		qos.transient_local();
		offboard_control_mode_publisher_ = this->create_publisher<OffboardControlMode>("/fmu/in/offboard_control_mode", qos);
		trajectory_setpoint_publisher_ = this->create_publisher<TrajectorySetpoint>("/fmu/in/trajectory_setpoint", qos);
		vehicle_command_publisher_ = this->create_publisher<VehicleCommand>("/fmu/in/vehicle_command", qos);
```

JetJie, 2022-11-17:
@laylow13 Could you share me the whole code? Obviously, you add qos snippet in the cpp example. Thanks in advance.