PX4/px4_ros_com #168: Could not compile micrortps_agent

laylow13, 2022-11-12:
The current version of the repository does not contain the compile options and source code of micrortps_agent and cannot compile micrortps_agent.

JetJie, 2022-11-14:
> The current version of the repository does not contain the compile options and source code of micrortps_agent and cannot compile micrortps_agent.

yes, in the latest version, they have removed the rtps; now it is based microdds. I am also urgent to find a usable cpp offboard example.

ripdk12, 2022-11-15:
I am also having the same problem, please update the documentation which obviously is completely outdated now.

TSC21, 2022-11-15:
Please be patient as the documentation is already undergoing an update: https://github.com/PX4/PX4-user_guide/pull/2094

berndpfrommer, 2022-12-26:
I have been using ROS2 for a while, also PX4 based controllers, but I am totally at a loss regarding the use of this repository. I cannot find any agent code. Something changed fundamentally. What is the last version that reasonably correlates with the documentation?

dagar, 2022-12-27:
> I have been using ROS2 for a while, also PX4 based controllers, but I am totally at a loss regarding the use of this repository. I cannot find any agent code. Something changed fundamentally. What is the last version that reasonably correlates with the documentation?

@berndpfrommer with current PX4 main/development all you need to do is use the regular micro-ros-agent (snap, docker, etc) and all the typical PX4 builds have "microdds_client" included by default that bridges everything. Feel free to ping me on slack or discord if you want any questions. This all needs to be documented properly before the v1.14 stable release.

berndpfrommer, 2022-12-27:
@dagar Thanks for responding so quickly.
It turns out for my platform (pixracer) the microdds_client is not built by default. That took me a while to figure out. For others struggling with this, you need to edit ``boards/px4/fmu-v4/default.px4board`` and add the following line: ``CONFIG_MODULES_MICRODDS_CLIENT=y``.

Plus of course I had to switch off mavlink, switch on the microdds_client in QGroundControl.
[This post was very helpful](https://github.com/PX4/PX4-Autopilot/issues/20634#issue-1455130160).

dagar, 2022-12-27:
> @dagar Thanks for responding so quickly. It turns out for my platform (pixracer) the microdds_client is not built by default. That took me a while to figure out. For others struggling with this, you [...]

If it fits I'd like to squeeze it in everywhere we can by default and make this the norm going forward. There's quite a bit more integration coming that will build on this.