PX4/px4_ros_com #171: Compatible version with PX4-Autopilot to use rtps

MineCold20, 2022-12-02:
Hi! I'm having some troubles to run `sensor_combined_listener.launch.py` node in "Sanity Check the Installation" section of [https://docs.px4.io/main/en/ros/ros2_comm.html](url). 

When I start the node no data is streamed in terminal. The following warning is printed in the terminal where micrortps_agent is running: `[   micrortps_agent   ]	Unexpected topic ID '62' to publish. Please make sure the agent is capable of parsing the message associated to the topic ID '62'`. 
When I installed PX4-Autopilot and px4_ros_com and px4_msgs packages, I was not able to install the last release because rtps doesn't work on newest version and documentation is not updated yet to make it works. So I think there's a compatibility problem between PX4-Autopilot and px4_ros_com and px4_msgs. 

Anyone can suggest me a pair of packages version compatible with each other?

beniaminopozzan, 2022-12-06:
Hi!
First of all, if you want to use the micrortps bridge you need to refer to the documentation for version 1.13
https://docs.px4.io/v1.13/en/ros/ros2_comm.html
In particular you need to

1. clone PX4-Autopilot from tag `v1.13.*` or from branch `release/1.13`.
2. clone the `release/1.13` branches of `px4_msgs` and `px4_ros_com`.

However, point 2 is still WIP, please refer to https://github.com/PX4/px4_ros_com/pull/170 for finding out the right commits and procedure to have everything working.

If instead you wish to use the latest version of PX4, then you need to use the microdds bridge, please refer to https://github.com/Jaeyoung-Lim/px4-offboard for an idea about how to set it up.

MineCold20, 2022-12-12:
Thank you for your reply. Following https://github.com/PX4/px4_ros_com/pull/170 I'm getting this error while building `px4_ros_com`: 

stderr: px4_ros_com                              
/usr/bin/ld: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/DebugArray_Subscriber.cpp.o: in function `eprosima::fastrtps::SampleInfo_t::SampleInfo_t()':
/opt/ros/foxy/include/fastrtps/subscriber/SampleInfo.h:46: undefined reference to `eprosima::fastrtps::rtps::Time_t::Time_t()'
/usr/bin/ld: /opt/ros/foxy/include/fastrtps/subscriber/SampleInfo.h:46: undefined reference to `eprosima::fastrtps::rtps::Time_t::Time_t()'
/usr/bin/ld: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/DebugKeyValue_Subscriber.cpp.o: in function `eprosima::fastrtps::SampleInfo_t::SampleInfo_t()':
/opt/ros/foxy/include/fastrtps/subscriber/SampleInfo.h:46: undefined reference to `eprosima::fastrtps::rtps::Time_t::Time_t()'
/usr/bin/ld: /opt/ros/foxy/include/fastrtps/subscriber/SampleInfo.h:46: undefined reference to `eprosima::fastrtps::rtps::Time_t::Time_t()'
/usr/bin/ld: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/DebugValue_Subscriber.cpp.o: in function `eprosima::fastrtps::SampleInfo_t::SampleInfo_t()':
/opt/ros/foxy/include/fastrtps/subscriber/SampleInfo.h:46: undefined reference to `eprosima::fastrtps::rtps::Time_t::Time_t()'
/usr/bin/ld: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/DebugValue_Subscriber.cpp.o:/opt/ros/foxy/include/fastrtps/subscriber/SampleInfo.h:46: more undefined references to `eprosima::fastrtps::rtps::Time_t::Time_t()' follow
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:2112: micrortps_agent] Error 1
make[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

Failed   <<< px4_ros_com [3.99s, exited with code 2]

Summary: 1 package finished [14.5s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com
not found

What I've done:

1. Execute the following commands: 
`git clone https://github.com/PX4/PX4-Autopilot.git`
`cd PX4-Autopilot`
`git checkout v1.13.2`
`git submodule update --init --recursive`
`ROS2_WS_DIR=~/px4_ros_com_ros2/ make update_ros2_bridge`
2. Download `px4_msgs` and `px4_ros_com` packages respectively from https://github.com/PX4/px4_msgs/commit/688a5720292ec420261dc0e9e11f2c9c0721b418 and https://github.com/PX4/px4_ros_com/commit/1562ff30d56b7ba26e4d2436724490f900cc2375. 
3. Modify msg/GpsInjectData.msg as in https://github.com/PX4/px4_ros_com/pull/170#issuecomment-1331449462
4. Build the packages with `source build_ros2_workspace.bash`

Did I make something wrong?

beniaminopozzan, 2022-12-26:
Hi @MineCold20 , did you solve the issue?
It seems closely related to https://github.com/PX4/px4_ros_com/issues/137 (which you solved yourself :)) )

Anyway, now the procedure is even more simplified as we have specific docs for `v1.13`:
https://docs.px4.io/v1.13/en/ros/ros2_comm.html
Follow it follow it thoroughly and use a *clean* ros2 workspace.

MineCold20, 2022-12-29:
Hi @beniaminopozzan. I've solved cleaning the .bashrc script from useless stuff. Thank you for your help!

closed 2022-12-29 by MineCold20