PX4/px4_ros_com #194: Possible wrong conversion in frame_transforms.h

mzahana, 2023-05-04:
Hi @TSC21 , @beniaminopozzan 

I noticed in function `px4_to_ros_orientation(const T &in)` [here](https://github.com/PX4/px4_ros_com/blob/0bcf68bcb635199adcd134e8932932054e863c0d/include/px4_ros_com/frame_transforms.h#L316-L319), the function `baselink_to_aircraft_orientation(ned_to_enu_orientation(in));` is used although it's a conversion from PX4 to ROS. Shouldn't the function `aircraft_to_baselink_orientation` be used instead of `baselink_to_aircraft_orientation`. The same thing for the function `ros_to_px4_orientation` [here](ros_to_px4_orientation).

Can you please double check ?
Thank you.

beniaminopozzan, 2023-05-04:
Formally speaking, given the quaternion `in` that rotates vectors given in FRD frame into NED frame (the input of `px4_to_ros_orientation(const T &in)`), to have the quaterion that rotates vectors given in FLU frame into ENU frame you have

- To right multiply `in` by a quaternion that rotates FLU into FRD. Why that? Because `in` wants a vector in FRD, but in the end you what to input something in FLU, then you need to rotate from FLU to FRD.
- To left multiply `in` by a quaternion that rotates NED into ENU. Why that? Because `in` returns a vector in NED, but in the end you what to have something in ENU, then you need to rotate from NED to ENU.

Putting all together you have
`out = NED_TO_ENU_Q * in * FLU_TO_FRD_Q`

Which is exactly what `px4_to_ros_orientation(const T &in)` does:
`baselink_to_aircraft_orientation(ned_to_enu_orientation(in))` = `baselink_to_aircraft_orientation(NED_TO_ENU_Q * in) ` = `NED_TO_ENU_Q * in * FLU_TO_FRD_Q`

Side note: the NED_TO_ENU and FLU_TO_FRD rotation are **special** as it holds that NED_TO_ENU=ENU_TO_NED and FLU_TO_FRD=FRD_TO_FLU

This is why you have this in the code
https://github.com/PX4/px4_ros_com/blob/0bcf68bcb635199adcd134e8932932054e863c0d/src/lib/frame_transforms.cpp#L157-L177

mzahana, 2023-05-04:
Thanks for the clarification!

closed 2023-05-04 by mzahana