PX4/px4_ros_com #195: How to transform angular velocity using frame_transfroms.h ?

mzahana, 2023-05-04:
Hi!

I would like to transform angular velocity vector from PX4 to ROS using the functions provided in [frame_transforms.h](https://github.com/PX4/px4_ros_com/blob/0bcf68bcb635199adcd134e8932932054e863c0d/include/px4_ros_com/frame_transforms.h) ?
Is [aircraft_to_baselink_body_frame](https://github.com/PX4/px4_ros_com/blob/0bcf68bcb635199adcd134e8932932054e863c0d/include/px4_ros_com/frame_transforms.h#L354) the right function to use?

beniaminopozzan, 2023-05-04:
Hi @mzahana !
Yes, it is! Just to be sure, you want to transform the angular velocity provided by which topic?

mzahana, 2023-05-04:
Yes, the angular velocity provided by vehicle odometry topic

mzahana, 2023-05-04:
Thanks @beniaminopozzan .

closed 2023-05-04 by mzahana