PX4/px4_ros_com #197: Debugging failure to switch to offboard

vranki, 2023-05-25:
I'm writing a offboard controller in c++. It's based on examples found in the px4_ros_com repository. The program can fly a quadcopter perfectly fine in simulator (gazebo + QGroundControl). But on real Pixhawk hardware it fails to switch to offboard randomly maybe 90% of tries. I'm looking for method of figuring out what's the cause. QGroundControl just says offboard was rejected, but doesn't provide any detail on actual reason.

I should be outputting the correct topics (trajectorysetpoint in position and velocity control modes) and the fact that it works always on simulator and sometimes on hardware also suggests this. Of course there is a chance of a bug but I'm stuck trying to find it.
The hardware itself seems healthy and flies fine using non-offboard control modes.

Any help is appreciated.

beniaminopozzan, 2023-05-25:
Hi @vranki 
Is the command rejected even if you send it from QGC ?
You could also subscribe to https://github.com/PX4/PX4-Autopilot/blob/main/msg/VehicleCommandAck.msg (add it to the bridge first) to have more info about what went wrong

vranki, 2023-05-26:
On real drone we have a transmitter switch to go to offboard mode, so it's not possible to control it from QGC. In simulator switching to offboard from QGC works always. I'll try subscribing to VehicleCommandAck and see if it gives any useful hints.