PX4/px4_ros_com #199: Confusing description & links to documentation

not7cd, 2023-06-14:
Hey, I got confused by the role of this package.
The sidebar description says: 
> ROS2/ROS interface with PX4 through a **Fast-RTPS** bridge 
While README says
> This package materializes the ROS2 side of the PX4 DDS bridge, establishing a bridge between the PX4 autopilot stack through an **XRCE-DDS** bridge. ...

Later it links to https://docs.px4.io/main/en/middleware/micrortps.html, which redirects to https://docs.px4.io/main/en/middleware/uxrce_dds.html. So I don't know if this package is still relevant. `px4_ros_com` appears 3 times on the later page without much context.

Also, README still links to the defunct Slack.

beniaminopozzan, 2023-06-14:
Hi @not7cd , yes, this repo requires some work as PX4 moved from _Fast-RTPS_ to _XRCE-DDS_.
The packages itself provides a library for PX4-ROS frame conversions and some example of ROS 2 - PX4 operations.

not7cd, 2023-06-14:
I think it would be best to reflect this in the project's description then. Maybe create a highlighted text pointing out the migration.