PX4/px4_ros_com #214: Error raised when executing build_ros2_workspace.bash

xirhxq, 2024-10-23:
Hello all!

I am trying to communicate to PX4 with a Jetson Orin NX on Ubuntu 20.04 & ROS2 Foxy. I was doing the steps according to [_Build ROS2 Workspace_ in PX4 Official Documentation](https://docs.px4.io/v1.13/en/ros/ros2_comm.html#build-ros-2-workspace.), and I got some problems.

Here is the output when executing `source build_ros2_workspace.sh`:

```bash
amov@amov:~/px4_ros_com_ros2/src/px4_ros_com/scripts$ ./build_ros2_workspace.bash
[0.580s] WARNING:colcon.colcon_core.package_selection:ignoring unknown package 'ros1_bridge' in --packages-skip
Starting >>> px4_msgs
Finished <<< px4_msgs [13.7s]                     
Starting >>> px4_ros_com
--- stderr: px4_ros_com                              
c++: error: /home/amov/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp: No such file or directory
c++: fatal error: no input files
compilation terminated.
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:497: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_agent.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< px4_ros_com [1.56s, exited with code 2]

Summary: 1 package finished [15.6s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com
```

And I add `--verbose` to it, and here is what I got:

```bash
amov@amov:~/px4_ros_com_ros2/src/px4_ros_com/scripts$ ./build_ros2_workspace.bash --verbose
FastRTPSGen version: 1.0
[0.581s] WARNING:colcon.colcon_core.package_selection:ignoring unknown package 'ros1_bridge' in --packages-skip
Starting >>> px4_msgs
[  0%] Built target px4_msgs__cpp            
[ 12%] Built target px4_msgs__rosidl_generator_c
[ 24%] Built target px4_msgs__rosidl_typesupport_cpp
[ 37%] Built target px4_msgs__rosidl_typesupport_introspection_cpp
[ 50%] Built target px4_msgs__rosidl_typesupport_c
[ 62%] Built target px4_msgs__rosidl_typesupport_fastrtps_cpp
[ 75%] Built target px4_msgs__rosidl_typesupport_introspection_c
[ 88%] Built target px4_msgs__rosidl_typesupport_fastrtps_c
[ 88%] Built target px4_msgs
[ 88%] Built target px4_msgs__py                 
[100%] Built target px4_msgs__python             
[100%] Built target px4_msgs__rosidl_typesupport_introspection_c__pyext
[100%] Built target px4_msgs__rosidl_typesupport_c__pyext
[100%] Built target px4_msgs__rosidl_typesupport_fastrtps_c__pyext
-- Install configuration: "RELWITHDEBINFO"        
-- Execute custom install script
-- Up-to-date symlink: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/ament_index/resource_index/rosidl_interfaces/px4_msgs
-- Up-to-date symlink: /home/amov/px4_ros_com_ros2/install/px4_msgs/include/px4_msgs/msg/action_request.h
-- Up-to-date symlink: /home/amov/px4_ros_com_ros2/install/px4_msgs/include/px4_msgs/msg/actuator_armed.h
...
...
~2000 of 'Up-to-date symlink' outputs
...
...
-- Up-to-date symlink: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/package.xml
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/lib/libpx4_msgs__rosidl_generator_c.so
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/lib/libpx4_msgs__rosidl_typesupport_introspection_c.so
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/lib/libpx4_msgs__rosidl_typesupport_c.so
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/lib/libpx4_msgs__rosidl_typesupport_introspection_cpp.so
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/lib/libpx4_msgs__rosidl_typesupport_cpp.so
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake/px4_msgs__rosidl_generator_cExport.cmake
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake/px4_msgs__rosidl_generator_cExport-relwithdebinfo.cmake
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake/px4_msgs__rosidl_typesupport_introspection_cExport.cmake
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake/px4_msgs__rosidl_typesupport_introspection_cExport-relwithdebinfo.cmake
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake/px4_msgs__rosidl_typesupport_cExport.cmake
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake/px4_msgs__rosidl_typesupport_cExport-relwithdebinfo.cmake
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake/px4_msgs__rosidl_generator_cppExport.cmake
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake/px4_msgs__rosidl_typesupport_introspection_cppExport.cmake
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake/px4_msgs__rosidl_typesupport_introspection_cppExport-relwithdebinfo.cmake
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake/px4_msgs__rosidl_typesupport_cppExport.cmake
-- Up-to-date: /home/amov/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake/px4_msgs__rosidl_typesupport_cppExport-relwithdebinfo.cmake
Finished <<< px4_msgs [13.6s]                    
Starting >>> px4_ros_com
[  1%] Generating micro-RTPS agent code...       
[  3%] Built target frame_transforms
[  7%] Built target offboard_control
[  7%] Built target debug_vect_advertiser
[  9%] Built target vehicle_gps_position_listener
[ 11%] Built target sensor_combined_listener
Traceback (most recent call last):                   
  File "/home/amov/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 178, in <module>
    fastrtpsgen_version_out = subprocess.check_output(
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/usr/local/bin/fastrtpsgen'
[ 11%] Generating micro-RTPS agent code...           
Traceback (most recent call last):                   
  File "/home/amov/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 178, in <module>
    fastrtpsgen_version_out = subprocess.check_output(
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/usr/local/bin/fastrtpsgen'
[ 12%] Building CXX object CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_agent.cpp.o
c++: error: /home/amov/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp: No such file or directory
c++: fatal error: no input files
compilation terminated.
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:497: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_agent.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
--- stderr: px4_ros_com                              
c++: error: /home/amov/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp: No such file or directory
c++: fatal error: no input files
compilation terminated.
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:497: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_agent.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< px4_ros_com [1.58s, exited with code 2]

Summary: 1 package finished [15.5s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com
```

Genuinely appreciate it if someone can help!

AlexisTM, 2024-10-23:
Are you on PX4 1.13? 
Since 1.14, it isn't using the micrortps agent, but uXRCE.

xirhxq, 2024-10-23:
> Are you on PX4 1.13? Since 1.14, it isn't using the micrortps agent, but uXRCE: https://docs.px4.io/v1.15/en/ros2/user_guide

Sorry I forgot to say that. I am on 1.13.